LETARTARE/Grbl-xx_with_Arduino #18: Another request, not an issue - Multi Stepper Driver option

Highcooley, 2015-01-05:
Hey, nice job converting grbl for mega2560

Is there a posibility to configure grbl to use two stepper drivers for one axis simultaneously (e.g. use E1 of a RAMPS board for Y), so two steppers of a portal CNC mill have individual drivers each and thus are able to use the full driver power? Thanks for our reply!

Keep your nice work up!

RoelandK, 2015-01-05:
You can just share the Driver and Step pin to 2 drivers... if needed add a
diode to make sure you dont have any feedback.

Highcooley, 2015-01-05:
Thanks for your reply! This would mean to modify the RAMPS hardware. Could it also be done in the firmware?

RoelandK, 2015-01-05:
Hmm like that of course.. i think it should be possible.

I cant look at the code right now. I will do so when I get back from work.
You want to use E1 as Y-2 right?

Kr,
Roeland

Highcooley, 2015-01-05:
Hi Roeland

Yes, E1 as Y-2 would be the idea. If E1 is the default driver for the 4th axis, also E2 would do. Of course, the best solution would be to make 4th axis as well as dual driver configurable for other people's needs (a variable for each axis which can be set to the preferred additional driver or -1 for off, for instance). Thank you very much!

Cheers,
Andy